espterm/espterm-firmware #263: How to adapt memory to 512K ESP01?

Karl-opec, 2018-09-23:
I tried to flash  a ESP-01, 512KB  with flash.sh but got the following message:

`A fatal error occurred: File 2.4.0-pre2-0x40000-en.bin (length 383568) at offset 262144 will not fit in 524288 bytes of flash. Use --flash-size argument, or change flashing address.`

How can i run the software on such a device?

MightyPork, 2018-09-23:
Honestly, I'm not sure. That must be one of the first generation ESP-01 boards, right?
They now all come with at least 1 MB of flash. I don't have any ESP-01 like this to test.

If there's a solution, it might involve editing `ldscript_memspecific.ld` and likely something else in the makefiles, as per https://github.com/pfalcon/esp-open-sdk/issues/226#issuecomment-265415461 and https://bbs.espressif.com/viewtopic.php?t=784 - then, of course, rebuild the firmware.

note `ldscript_memspecific.ld` overrides the mentioned `eagle.app.v6.ld` file.

Karl-opec, 2018-09-29:
Thank you for your hints,

I managed to get it compiled and flashed it to 0x20000.
Modified Makefile.combined:

`ldscript_memspecific.ld:
        $(vecho) "GEN $@"
        $(Q) echo "MEMORY { irom0_0_seg : org = 0x40220000, len = "$$(printf "0x%X" $$(($(ESP_SPI_FLASH_SIZE_K)*1024-0x24000)))" }"> ldscript_memspecific.ld`
and changed eagle.app.v6.ld appropriate.

On the debug uart i get the following output.

```
[i] ================ ESPTerm ================
[i] 
[i] Project by Ondrej Hruska, 2017-2018
[i] 
[i] github.com/espterm/espterm-firmware
[i] 
[i] Version 2.4.0-pre2 (en), code name "Damselfly"
[i]  back-end #e125aec front-end #9aae724
[i]  built Sep 29 2018 at 19:34:20 CEST
[i] 
[i] Device ID: 999E23

00:00:00 [ ] Note: Hold GPIO0 low for reset to AP mode.
mode : null
00:00:00 [i] [Persist] Loading settings from FLASH...
00:00:00 [ ] Persist memory map:
00:00:00 [ ] > wifi  at    0 (error  0)
00:00:00 [ ] > sys   at  340 (error  0)
00:00:00 [ ] > term  at  640 (error  0)
00:00:00 [ ] > cksum at 1896 (error  0)
00:00:00 [ ] > Total size = 1900 bytes (error 0)
00:00:00 [i] [Persist] Checksum verification: PASSED
00:00:00 [ ] [Persist] Applying live settings...
00:00:00 [ ] [Persist] > system
00:00:00 [i] COM SERIAL: 115200 baud, NONE parity, 1 stopbit(s)
00:00:00 [ ] [Persist] > wifi
mode : sta(18:fe:34:99:9e:23) + softAP(1a:fe:34:99:9e:23)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
bcn 0
del if1
add if1
d00:00:01 [ ] Rx/Tx: 0/0%, Hp: 22144 (+22144)
00:00:06 [ ] Rx/Tx: 0/0%, Hp: 22144
chg_A4:0
00:00:11 [ ] Rx/Tx: 0/0%, Hp: 22144
00:00:16 [ ] Rx/Tx: 0/0%, Hp: 22144
```

`add 1
aid 1
station: xx:xx:xx:xx:xx:xx join, AID = 1
00:04:05 [ ] Rx/Tx: 0/0%, Hp: 21824 (-320)
`
The access point is open and i can connect a wifi client. 
this client gets ip 192.168.4.100 and tries to shows a capture page.

But nothing is shown, only html error 404
then i can disconnect the client from ap and i get following:
`station: 48:4b:aa:a0:31:e2 leave, AID = 1
rm 1
`
So far it seems running bat where are the html pages. I already have 
made make web. it has generated the html subfolder in source.

Any Idea how to procedd from here?

MightyPork, 2018-09-29:
That's good progress!
I'll try to replicate your changes and see if I get the same result. The log output looks perfectly fine. Maybe the "filesystem" module is looking at the wrong address or something like that

MightyPork, 2018-10-06:
Okay, so I gave it a try, but unfortunately I can't replicate your success (at least using my 4MB device). 

- After modifying the LD script to use 0x20000 instead of 0x40000, it became unstable and e.g. crashed when I tried to open settings. But the web pages loaded OK. 
- However, after I also set my ESP_SPI_FLASH_SIZE_K to 512 in esphttpdconfig.mk (it was 1024 before), it failed to build with the "does not fit" error. 
- After changing `$(ESP_SPI_FLASH_SIZE_K)*1024-0x24000` to $(ESP_SPI_FLASH_SIZE_K)*1024-0x4000`, it built, but resulted in a bootloop.

I got it to show me the 404 page once, but I'm unable to replicate it. If you get it to work, let me know, but I'm afraid I'm unable to help here.